freakent/dbus-mqtt-devices #137: Problems after Update to VebusOS V3.50-3.51

rholze, 2024-11-17:
### Version

0.8.0 (current)

### VenusOS version

V3.51

### Does this issue occur during installation or at runtime ?

Runtime

### Is there a Discussion thread for this issue? Please paste a link.

https://community.victronenergy.com/t/after-update-to-3-50-i-could-not-install-any-nodes-in-node-red/10014

### What is the issue you are actually experiencing?

After creating the thread in the victron community forum, I have now carried out a few tests. As soon as I installed the dbus-mqtt-device driver, I can't install any plug-ins on the same machine via the palette in node-red. The behavior is also different whether you have updated from an older version of venusOS or flashed a fresh SD card (in the victron forum was the time of the update from an older version of venusOS!). With an update, not a single plug-in can be installed. With a completely new SD card, some plug-ins work, others don't?!? I don't have the know-how to go deeper into the error. Can you reproduce the problems?
Thanks Ronald

### Relevant log output

```shell
0 verbose cli /usr/bin/node /usr/bin/npm
1 info using npm@10.7.0
2 info using node@v18.20.3
3 silly config:load:file:/usr/lib/node_modules/npm/npmrc
4 warn config production Use `--omit=dev` instead.
5 silly config:load:file:/data/home/nodered/.node-red/.npmrc
6 silly config:load:file:/data/home/nodered/.npmrc
7 silly config:load:file:/usr/etc/npmrc
8 verbose title npm install node-red-contrib-uibuilder@7.0.4
9 verbose argv "install" "--no-audit" "--no-update-notifier" "--no-fund" "--save" "--save-prefix" "~" "--production" "--engine-strict" "node-red-contrib-uibuilder@7.0.4"
10 verbose logfile logs-max:10 dir:/data/home/nodered/.npm/_logs/2024-11-08T16_45_14_547Z-
11 verbose logfile /data/home/nodered/.npm/_logs/2024-11-08T16_45_14_547Z-debug-0.log
12 silly logfile done cleaning log files
13 silly idealTree buildDeps
14 silly fetch manifest node-red-contrib-uibuilder@7.0.4
15 http fetch GET 200 https://registry.npmjs.org/node-red-contrib-uibuilder 1285ms (cache revalidated)
16 silly placeDep ROOT node-red-contrib-uibuilder@7.0.4 OK for: node-red-project@0.0.1 want: 7.0.4
17 silly fetch manifest @totallyinformation/ti-common-event-handler@^1.0.0
18 silly fetch manifest degit@^2.8.4
19 silly fetch manifest express@^4.17.1
20 http fetch GET 200 https://registry.npmjs.org/express 365ms (cache revalidated)
21 silly fetch manifest fast-glob@^3.2.7
22 http fetch GET 200 https://registry.npmjs.org/degit 899ms (cache revalidated)
23 silly fetch manifest fs-extra@^11.1.0
24 http fetch GET 200 https://registry.npmjs.org/fs-extra 168ms (cache revalidated)
25 silly fetch manifest jsdom@^25.0.0
26 http fetch GET 200 https://registry.npmjs.org/jsdom 168ms (cache revalidated)
27 silly fetch manifest socket.io@^4.3.2
28 http fetch GET 200 https://registry.npmjs.org/socket.io 166ms (cache revalidated)
29 http fetch GET 200 https://registry.npmjs.org/fast-glob 2519ms (cache miss)
30 http fetch GET 200 https://registry.npmjs.org/@totallyinformation%2fti-common-event-handler 3184ms (cache miss)
31 silly fetch manifest canvas@^2.11.2
32 http fetch GET 200 https://registry.npmjs.org/canvas 1366ms (cache miss)
33 silly placeDep node_modules/node-red-contrib-uibuilder @totallyinformation/ti-common-event-handler@ OK for: node-red-contrib-uibuilder@7.0.4 want: ^1.0.0
34 silly placeDep ROOT degit@2.8.4 OK for: node-red-contrib-uibuilder@7.0.4 want: ^2.8.4
35 silly placeDep ROOT express@4.21.1 OK for: node-red-contrib-uibuilder@7.0.4 want: ^4.17.1
36 silly placeDep node_modules/node-red-contrib-uibuilder fast-glob@ OK for: node-red-contrib-uibuilder@7.0.4 want: ^3.2.7
37 silly placeDep ROOT fs-extra@11.2.0 OK for: node-red-contrib-uibuilder@7.0.4 want: ^11.1.0
38 silly placeDep ROOT jsdom@25.0.1 OK for: node-red-contrib-uibuilder@7.0.4 want: ^25.0.0
39 silly placeDep ROOT socket.io@4.8.1 OK for: node-red-contrib-uibuilder@7.0.4 want: ^4.3.2
40 silly fetch manifest qs@6.13.0
41 silly fetch manifest depd@2.0.0
42 silly fetch manifest etag@~1.8.1
43 http fetch GET 200 https://registry.npmjs.org/qs 177ms (cache revalidated)
44 silly fetch manifest send@0.19.0
45 http fetch GET 200 https://registry.npmjs.org/depd 721ms (cache miss)
[... 257 lines cut ...]
303 silly placeDep ROOT punycode@2.3.1 OK for: tr46@5.0.0 want: ^2.3.1
304 verbose type system
305 verbose stack FetchError: Invalid response body while trying to fetch https://registry.npmjs.org/@totallyinformation%2fti-common-event-handler: Unknown system error -74: Unknown system error -74, mkdir '/data/home/nodered/.npm/_cacache/content-v2/sha512/d5/cb'
305 verbose stack     at /usr/lib/node_modules/npm/node_modules/minipass-fetch/lib/body.js:170:15
305 verbose stack     at async Response.json (/usr/lib/node_modules/npm/node_modules/minipass-fetch/lib/body.js:75:17)
305 verbose stack     at async RegistryFetcher.packument (/usr/lib/node_modules/npm/node_modules/pacote/lib/registry.js:98:25)
305 verbose stack     at async RegistryFetcher.manifest (/usr/lib/node_modules/npm/node_modules/pacote/lib/registry.js:129:23)
306 verbose cwd /data/home/nodered/.node-red
307 verbose Linux 5.10.110-rpi-venus-4
308 verbose node v18.20.3
309 verbose npm  v10.7.0
310 error code EEXIST
311 error syscall mkdir
312 error path /data/home/nodered/.npm/_cacache/content-v2/sha512/d5/cb
313 error errno EEXIST
314 error Invalid response body while trying to fetch https://registry.npmjs.org/@totallyinformation%2fti-common-event-handler: Unknown system error -74: Unknown system error -74, mkdir '/data/home/nodered/.npm/_cacache/content-v2/sha512/d5/cb'
315 error File exists: /data/home/nodered/.npm/_cacache/content-v2/sha512/d5/cb
316 error Remove the existing file and try again, or run npm
317 error with --force to overwrite files recklessly.
318 verbose exit 1
319 verbose code 1
320 silly unfinished npm timer reify 1731084318527
321 silly unfinished npm timer reify:loadTrees 1731084318547
322 error A complete log of this run can be found in: /data/home/nodered/.npm/_logs/2024-11-08T16_45_14_547Z-debug-0.log
```

freakent, 2024-11-17:
Hi, there are a few errors that look like this.
`305 verbose stack FetchError: Invalid response body while trying to fetch https://registry.npmjs.org/@totallyinformation%2fti-common-event-handler: Unknown system error -74: Unknown system error -74, mkdir '/data/home/nodered/.npm/_cacache/content-v2/sha512/d5/cb'
`
Have you run out of filesystem storage space?
From ssh type 
`df -h`
And post the results.

rholze, 2024-11-18:
Hello, I have two systems to test. One (the first one in the listing) on which I have made an update, and a second one on which I have installed a new image. I had already tried to research error -74 on the Internet, but found nothing. I suspected an authorization problem or moved folders, but didn't get any further.

Is there a script that I can use to completely uninstall the driver? Then I could test whether it can then be reinstalled on the updated system?

Thanks Ronald

************ system with update ************
login as: root
root@192.168.178.81's password:
Last login: Thu Nov 14 16:31:49 2024
root@raspberrypi2:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 8.9G      1.1G      7.4G  13% /
devtmpfs                428.6M      4.0K    428.6M   0% /dev
tmpfs                   461.1M    848.0K    460.2M   0% /run
tmpfs                   461.1M    368.0K    460.7M   0% /var/volatile
/dev/mmcblk0p1           41.2M     21.8M     19.4M  53% /u-boot
/dev/mmcblk0p4            9.6G    135.5M      9.0G   1% /data
tmpfs                   461.1M    848.0K    460.2M   0% /service
root@raspberrypi2:~#

************** fresh system ***************
Using username "root".
root@venus.local's password:
Last login: Sun Nov 17 11:59:27 2024 from 2003:ee:973d:d500:58f4:2a13:14a7:b7f1
root@raspberrypi2:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 2.3G      1.1G      1.1G  50% /
devtmpfs                428.6M      4.0K    428.6M   0% /dev
tmpfs                   461.1M    860.0K    460.2M   0% /run
tmpfs                   461.1M    524.0K    460.6M   0% /var/volatile
/dev/mmcblk0p1           41.2M     21.9M     19.4M  53% /u-boot
/dev/mmcblk0p4            2.4G    101.2M      2.2G   4% /data
tmpfs                   461.1M    860.0K    460.2M   0% /service
root@raspberrypi2:~#

rholze, 2024-11-18:
After I uninstalled the drivers and then tried to install the dashboard in node-red, I got the following error:
```
-----------------------------------------------------------
2024-11-18T12:27:45.156Z Installieren : node-red-dashboard 3.6.5

2024-11-18T12:27:45.024Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-dashboard@3.6.5
2024-11-18T12:27:46.982Z [err] npm warn config production Use `--omit=dev` instead.
2024-11-18T12:28:18.601Z [err] npm error code EEXIST
2024-11-18T12:28:18.603Z [err] npm error syscall mkdir
2024-11-18T12:28:18.603Z [err] npm error path /data/home/nodered/.npm/_cacache/content-v2/sha512/cc/ee
2024-11-18T12:28:18.604Z [err] npm error errno EEXIST
2024-11-18T12:28:18.630Z [err] npm error Invalid response body while trying to fetch https://registry.npmjs.org/compression: Unknown system error -74: Unknown system error -74, mkdir '/data/home/nodered/.npm/_cacache/content-v2/sha512/cc/ee'
2024-11-18T12:28:18.630Z [err] npm error File exists: /data/home/nodered/.npm/_cacache/content-v2/sha512/cc/ee
2024-11-18T12:28:18.631Z [err] npm error Remove the existing file and try again, or run npm
2024-11-18T12:28:18.633Z [err] npm error with --force to overwrite files recklessly.
2024-11-18T12:28:18.645Z [err] 
2024-11-18T12:28:18.646Z [err] npm error A complete log of this run can be found in: /data/home/nodered/.npm/_logs/2024-11-18T12_27_46_560Z-debug-0.log
2024-11-18T12:28:18.691Z rc=1
```
It is also not clear to me how the driver is connected to this? But I had a bare VenusOS installed. I was able to install and uninstall everything in node-red. After I installed the driver for dbus-mqtt-devices, it didn't work anymore?!??
Maybe someone, who has time, can recreate this? Or am I doing something wrong?
Regards Ronald

freakent, 2024-11-18:
I think the fact that you get this error after installing this driver is a coincidence. The driver does not use npm, it 's a purely python code base.

Googling this error I found this post on Stack Overflow with various tips of dealing with the npm cache.
https://stackoverflow.com/questions/64308304/npm-err-invalid-response-body-while-trying-to-fetch-http-registry-npmjs-org-a