freakent/dbus-mqtt-devices #141: Temp Sensors Stop Updating

pbawesome, 2024-12-12:
### Version

0.8.0 (current)

### VenusOS version

3.52

### Does this issue occur during installation or at runtime ?

Runtime

### What is the issue you are actually experiencing?

After some time the temperature sensors stop updating, they are registered, and the messages sent to mqtt-out node are correct.
Attached are the flows for registration, and temperature values that are being sent. Provided below are the message payloads for the Send RIX Temp Node

I also tried adding a function node in between and creating a new message with only payload and topic, no luck.

I rebooted, disconnected, and re-registered and that seems to work, until the next time it stops updating. Is there anything I can check? 

For now I put a timed inject to disconnect every hour, delay 10 seconds, and re-register. We'll see if that makes it more reliable.

[flows (1).json](https://github.com/user-attachments/files/18105974/flows.1.json)
[flows.json](https://github.com/user-attachments/files/18105975/flows.json)

### If this is not installation related, what does the status payload look like that you are publishing to the driver to register your device?

12/11/2024, 10:35:48 PM[node: debug 107](https://192.168.50.101:1881/#)
W/c0619ab413ee/temperature/1/Temperature : msg : Object
object
payload: object
topic: "W/c0619ab413ee/temperature/1/Temperature"
_msgid: "7890f240357bad7e"
_event: "node:b4bd43d5473313ea"
statusCode: 200
headers: object
responseUrl: "http://10.10.10.10/status.xml"
redirectList: array[0]
retry: 0
temperature: 18.6

12/11/2024, 10:43:41 PM[node: debug 107](https://192.168.50.101:1881/#)
W/c0619ab413ee/temperature/1/Temperature : msg : Object
object
payload: object
value: 18.5
topic: "W/c0619ab413ee/temperature/1/Temperature"
_msgid: "b577e5c2aebbba90"

freakent, 2024-12-12:
Ok, a couple of things
1) this is not something anyone has reported before nor have I encountered it. 

2) once a device is registered and you have a device id this driver is no longer involved, it's purely Victron's api you are working with. 

3) you haven't said what's running where. What's actually providing the temperature values when this happens?

4) does the GX device think this temperature sensor is disconnected? In your MQTT config are you sending a last will message to send a disconnect message when connection is broken?

If you check the Troubleshooting section of the README I have included some instructions on how to check the dbus-Mqtt-devices log file. See if you have any errors. 

I highly recommend using MQTT Explorer to monitor the MQTT topics involved. 

(Sorry if some of these questions are answered in the flows, but I'm on my mobile for most of today)

freakent, 2024-12-12:
Another thought, have you tried running the node red example in the samples directory?

pbawesome, 2024-12-12:
1. yeah I figured lol
2. Fair enough
3. I send an HTTP GET to retrieve an XML file, parse and extract the temperature value, send it into a change node (just like the example) to set msg.payload and msg.topic As you can see in the payloads above it is working fine. Perhaps I need to put a filter node so I only continue if I get a 200 response, maybe sending a null value breaks it until it is next disconnected?
4. It does not think it's disconnected, it just says last updated ## hours ago

Will look into the troubleshooting and mqtt explorer

I imported your example and then modified it to add my sensors.

pbawesome, 2024-12-12:
Excuse me for being a bit noobish, MQTT Explorer are you talking about this? https://mqtt-explorer.com/
What's the best way to run this on the effectively headless Cerbo GX?

freakent, 2024-12-12:
That's the correct MQTT explorer. You don't run it on your Cerbo you run it on your laptop. Point MQTT Explorer at the MQTT server on the Cerbo by using the Cerbo's IP address or try Venus.local. 

If you ssh into the Cerbo, there is also a Victron supplied utility called dbus-spy that show's you the actual values on the dbus.

pbawesome, 2024-12-12:
[image]
got it. I'll try this next time it wants to not update.

pbawesome, 2024-12-13:
on the MQTT side it's fine

I was looking at vrmlogger today for a separate project and happened to catch this.

`MainThread-dbusmonitor: com.victronenergy.temperature.mqtt_venusnr_temp01 disappeared from the dbus. Removing it from our lists
MainThread-dbusmonitor: com.victronenergy.temperature.mqtt_venusnr_temp02 disappeared from the dbus. Removing it from our lists
MainThread-dbusmonitor: com.victronenergy.temperature.mqtt_venusnr_temp03 disappeared from the dbus. Removing it from our lists
MainThread-dbusmonitor: com.victronenergy.temperature.mqtt_venusnr_temp04 disappeared from the dbus. Removing it from our lists
MainThread-dbusmonitor: com.victronenergy.temperature.mqtt_venusnr_temp05 disappeared from the dbus. Removing it from our lists
MainThread-dbusmonitor: com.victronenergy.temperature.mqtt_venusnr_temp06 disappeared from the dbus. Removing it from our lists
`
[image]
I do see it in dbus-spy missing a temperature

here is the node red input to the mqtt broker
`12/13/2024, 12:38:28 PM[node: debug 107](https://192.168.50.101:1881/#)
W/c0619ab413ee/temperature/1/Temperature : msg : Object
object
payload: object
value: 20.5
topic: "W/c0619ab413ee/temperature/1/Temperature"
_msgid: "c9981abba1743201"`

Here is MQTT explorer showing that it is correctly publishing to that topic
[image]

dbus mqtt logs

`2024-12-13 17:33:27.803301500 INFO:device_service:Unregistered mqtt_venusnr_temp01 from dbus
2024-12-13 17:33:27.803310500 INFO:device:Removed Service temp01 from client venusnr
2024-12-13 17:33:27.815851500 INFO:device_service:Unregistered mqtt_venusnr_temp02 from dbus
2024-12-13 17:33:27.817087500 INFO:device:Removed Service temp02 from client venusnr
2024-12-13 17:33:27.839100500 INFO:device_service:Unregistered mqtt_venusnr_temp03 from dbus
2024-12-13 17:33:27.840504500 INFO:device:Removed Service temp03 from client venusnr
2024-12-13 17:33:27.867223500 INFO:device_service:Unregistered mqtt_venusnr_temp04 from dbus
2024-12-13 17:33:27.868467500 INFO:device:Removed Service temp04 from client venusnr
2024-12-13 17:33:27.903395500 INFO:device_service:Unregistered mqtt_venusnr_temp05 from dbus
2024-12-13 17:33:27.904811500 INFO:device:Removed Service temp05 from client venusnr`

freakent, 2024-12-13:
The log message device_service:Unregistered will appear if the driver receives a disconnect message (connected = 0). If you are using nodered and have last will set up then nodered is sending the last will whenever you press the deploy button (because it resets its MQTT broker connections). Are you sending the register payload (connected =1) every time you redeploy your flow?

freakent, 2025-04-08:
Nothing heard for a while now.

closed 2025-04-08 by freakent