freakent/dbus-mqtt-devices #146: Not working since 3.60 beta

jeroenkl, 2025-04-06:
### Version

0.8.0 (current)

### VenusOS version

3.60 >betav39

### Does this issue occur during installation or at runtime ?

Runtime

### What is the issue you are actually experiencing?

https://community.victronenergy.com/t/v3-60-39-dbus-mqtt-devices-not-working-scarthgap-packages-unavailable/24869/44

### Relevant log output

```shell

```

jeroenkl, 2025-04-08:
Works! Thanks!

gharshaw, 2025-04-12:
@freakent , Just updated to -0.9.0-rc2, VenusOS 3.60~58.
I also ran /opt/victronenergy/swupdate-scripts/set-feed.sh candidate and opkg update to get the python dependencies 

And I'm getting this error. Any suggestions for a fix?

 *** starting dbus-mqtt-devices ***
@4000000067fac0f40f51e5dc dbus_mqtt_devices v{}
@4000000067fac0f40f51fd4c INFO:logger:Loglevel set to INFO
@4000000067fac0f40f692ecc INFO:dbus_mqtt_devices:-------- dbus_mqtt_devices, v is starting up --------
@4000000067fac0f410b828b4 INFO:device_manager:Using portalId b827eb54f79c
@4000000067fac0f431269694 Traceback (most recent call last):
@4000000067fac0f431287edc   File "/data/drivers/dbus-mqtt-devices-0.9.0-rc2/dbus_mqtt_devices.py", line 83, in <module>
@4000000067fac0f4313a2e34     main()
@4000000067fac0f4313e7fac   File "/data/drivers/dbus-mqtt-devices-0.9.0-rc2/dbus_mqtt_devices.py", line 65, in main
@4000000067fac0f4314c52bc     handler = MQTTDeviceManager(
@4000000067fac0f4314f38ec               ^^^^^^^^^^^^^^^^^^
@4000000067fac0f4314ff084   File "/data/drivers/dbus-mqtt-devices-0.9.0-rc2/device_manager.py", line 43, in __init__
@4000000067fac0f4315c11fc     MqttGObjectBridge.__init__(self, mqtt_server, CLIENTID, ca_cert, user, passwd, debug)
@4000000067fac0f43163af34   File "/data/drivers/dbus-mqtt-devices-0.9.0-rc2/ext/dbus-mqtt/mqtt_gobject_bridge.py", line 25, in __init__
@4000000067fac0f4316f0974     self._client = paho.mqtt.client.Client(client_id)
@4000000067fac0f43176d58c                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
@4000000067fac0f43177910c   File "/usr/lib/python3.12/site-packages/paho/mqtt/client.py", line 766, in __init__
@4000000067fac0f431af0074     raise ValueError(
@4000000067fac0f431b040c4 ValueError: Unsupported callback API version: version 2.0 added a callback_api_version, see migrations.md for details

Regards,
Gene

freakent, 2025-04-12:
Hi Gene,
You don't need to manually run /opt/victronenergy/swupdate-scripts/set-feed.sh and opkg update, the setup.sh does all that for you. 

I wonder if you somehow still have the soft links to Victron's version of the dbus-Mqtt library. Can you do an 
ls -l /data/drivers/dbus-mqtt-devices-0.9.0-rc2/ext ?

Martin

gharshaw, 2025-04-12:
***@***.***:~# ls -l /data/drivers/dbus-mqtt-devices-0.9.0-rc2/ext
lrwxrwxrwx    1 root     root            28 Apr 12 19:36 dbus-mqtt -> /opt/victronenergy/dbus-mqtt
lrwxrwxrwx    1 root     root            54 Apr 12 19:36 velib_python -> /opt/victronenergy/dbus-digitalinputs/ext/velib_python
***@***.***:~# 

Sent from my iPhone

> On Apr 12, 2025, at 3:50 PM, Martin Jarvis ***@***.***> wrote:
> 
> ls -l /data/drivers/dbus-mqtt-devices-0.9.0-rc2/ext

freakent, 2025-04-13:
@gharshaw My apologies, I just re-built the release. Please delete the rc2 version that you have and re-install.

rm -rf /data/drivers/dbus-mqtt-devices-0.9.0-rc2

gharshaw, 2025-04-14:
> [@gharshaw](https://github.com/gharshaw) My apologies, I just re-built the release. Please delete the rc2 version that you have and re-install.
> 
> rm -rf /data/drivers/dbus-mqtt-devices-0.9.0-rc2

@freakent 
Hi Martin, I appreciate the quick response. This fixed my issue.
Thank you,
Gene

derFrickler, 2025-06-05:
Thanks a lot, works just fine ;_)

freakent, 2025-06-12:
Issue fixed in release v0.9.0, see #161 for more details.