freakent/dbus-mqtt-devices #150: Reinstall needed after VenusOS update

Bart-1992, 2025-04-10:
### Version

0.8.0 (current)

### VenusOS version

3.55

### Does this issue occur during installation or at runtime ?

Other

### What is the issue you are actually experiencing?

First of all thank you very much for this module! Really appreciate your effort/work. 

I experience some issues when updating my venusOS to the next stable version ( in this case 3.55 ), but earlier updates resulted in same behaviour. 

I need to fully reinstall using your docs after performing VenusOS update, only after that the system works again as intended.
So for me it's not persistent across updates, as mentioned in your docs. 

Iooking at : 

```
root@raspberrypi4:~# svstat /service/dbus-mqtt-devices
/service/dbus-mqtt-devices: unable to chdir: file does not exist
```
It appears not to be installed anymore after the update?

Is there anything I can do about this? 

### Relevant log output

```shell

```

freakent, 2025-04-10:
This is very odd because I have tested this scenario many times and it works faultlessly for me. The /data folder is not erased during OS upgrades so that's my first concern. 

We need to understand what's different about your set up. 
What device are you using?
Are you installing into /data/drivers ?
Have you checked the contents of /data/rc.local?

There are more tips in the trouble shooting section of the README.

freakent, 2025-04-11:
Moving this to discussions until we identify the issue.